pqrs-org/Seil #45: anker bluetooth ios keyboard remap

chaitu236, 2015-01-01:
Hi,

I recently got this (http://www.ianker.com/product/98ANSLM78-WBTA) bluetooth keyboard. However, it is an ios/android keyboard and so, the default functionality for *Esc* button is *Home* and *Del* button is *Lock*. To use the *Esc* or *Del* keys, 'fn' key is to be used as a modifier.

I'm trying to remap the keys so that pressing the *Esc* (without fn) will send *Esc* event, and pressing *Del* (without fn) will send *Del*. There is currently no option for the same on Seil. I've tried Event Viewer of Karabiner but pressing the *Esc/Del* keys (without fn modifier) generate no event.

**Is there a way to map the keys??**

BTW I checked the kernel events sent by the keyboard in my android phone and the event code for *Esc* is 00ac, *fn+Esc* is 0001, *Del* is 0074, *fn+Del* is 006f.

tekezo, 2015-01-02:
Unfortunately, Seil and Karabiner does not support Home key in keyboards for iOS.
Seil and Karabiner use generic OS X keyboard driver and it ignores the Home key.

ControllerMate provides custom keyboard driver and it might support the Home key.
It's worth to try.
http://www.orderedbytes.com/controllermate/

closed 2015-01-02 by tekezo

chaitu236, 2015-01-02:
Thanks a lot!

kyosold, 2015-10-12:
@chaitu236  I had got this problem neither, did you fix this ??

chaitu236, 2015-10-12:
No I didnt.. Im just using it as is..

kyosold, 2015-10-13:
@chaitu236 I think this is not fix it, because i write code the app for OSX, this app to monitor keyboard event, i found that OSX can't get the HOME and DEL button event when i press down, but others button it can be catch event. so i think when i press the Home button, this keyboard can't send signal to macbook. so we can press FN+Home to instead of ESC. but this is uncomfortable use VIM ...

chaitu236, 2015-10-13:
@kyosold As tekezo already mentioned, the existing OSX driver discards home key.. so apps won't get the event... you'd need to use a different keyboard driver.... do you mean you already tried ControllerMate and even that doesn't work??

kyosold, 2015-10-16:
@chaitu236 I'm never used the ControllerMate, but i think same like this one, i give up, i'm learning to use FN+ESC, or use the physic fix, example: connect FN and ESC Key use wire in the keyboard T_T

tekezo, 2015-10-16:
If you don't use \` key, you can swap escape and \` key by Karabiner.
[image]

Or changing control-[ to escape is another alternative.
[image]

kyosold, 2015-10-16:
Yes, This is good idea, I had edit my Karabiner private.xml to only when use bluetooth keyboard and vim application, change ` to CTRL+[, this is my private.xml:

<?xml version="1.0"?>
<root>
	<devicevendordef>
		<vendorname>APPLE_PACKARD</vendorname>
		<vendorid>0x05ac</vendorid>
	</devicevendordef>
	
	<deviceproductdef>
		<productname>BLUETOOTH_KEYBOARD</productname>
		<productid>0x0239</productid>
	</deviceproductdef>
	
	
	<appdef>
		<appname>Vim</appname>
		<equal>com.vandyke.SecureCRT</equal>
	</appdef>
	
	<item>
		<name>Bluetooth 3.0 Keyboard Escape For Vim</name>
		<identifier>private.bluetooth_3.0_keyboard_escape_for_vim</identifier>

		<device_only>DeviceVendor::APPLE_PACKARD, DeviceProduct::BLUETOOTH_KEYBOARD</device_only>
		<only>Vim</only>

		<appendix>Change ` to ESC</appendix>
		<autogen>
			__KeyToKey__ 
			KeyCode::BACKQUOTE, 
			KeyCode::BRACKET_LEFT, ModifierFlag::CONTROL_L
		</autogen>
	</item>
</root>

NOTE: Maybe you need to change vendorid or productid, these value can be get from Karabiner EventViewer and example url :https://pqrs.org/osx/karabiner/xml.html.en