pqrs-org/Seil #46: How to map CapsLock as Escape AND Ctrl

lmartins, 2015-04-20:
My goal was to map CapsLock to Escape if pressed by itself and to Control if pressed in combination with another keys. I've heard somewhere this was possible to achieve with Seil but Im not able to figure it out. 

Could you help please?

tekezo, 2015-04-20:
You should use Karabiner.
https://pqrs.org/osx/karabiner/index.html.en

Change your caps lock key to control from System Preferences > Keyboard > Modifier Keys.
Then enable this setting in Karabiner Preferences.

[image]

(If you do not want to change the physical left control key, let me know.)

lmartins, 2015-04-20:
Brilliant, that was it. Thanks.

closed 2015-04-20 by lmartins