pqrs-org/Seil #47: Settings not persistent

justmytwospence, 2015-05-28:
I'm on Seli 11.1.0

tekezo, 2015-05-28:
I cannot reproduce this issue.
Are Seil settings cleared each restart?
Or you have to launch Seil manually?

justmytwospence, 2015-05-28:
I have to launch Seil manually, then reconfigure everything as if I had just installed Seil. I'm on OSX 10.9.5

tekezo, 2015-05-28:
Seil registers itself to Login Items automatically.
Please confirm Seil exists in your Login Items.
(You can confirm from System Preferences > Users & Groups >  Login Items.)

If Seil exists, do you press shift key while logged in?
If so, the shift key disables login items. Please release shift key until Seil is launched.
http://lifehacker.com/5795904/temporarily-disable-your-macs-login-items-with-the-shift-key

justmytwospence, 2015-05-28:
Can confirm Seil is a login item. I don't press the Shift key when logging in.

tekezo, 2015-05-28:
I see.
The cause might be either "Seil is terminated" or "login items does not work properly".

Please check system log. (You can confirm from Console.app.)
https://pqrs.org/osx/karabiner/faq.html.en#general

If there is any crash message, Seil is terminated at login.
Otherwise, I guess login items does not work properly.

If you opened too many document at shutdown, the login process restores these document and login items are delayed.
If so, please wait a minute at login for login items kick Seil, or open Seil manually.

justmytwospence, 2015-05-28:
This is what I see when grepping Console.app for `seil`:

```
5/27/15 6:56:38.089 PM com.apple.launchd.peruser.501[318]: (org.pqrs.Seil.10944[1062]) Spawned and waiting for the debugger to attach before continuing...
5/27/15 6:56:38.107 PM Seil[1062]: kextload
5/27/15 6:56:39.000 PM kernel[0]: org.pqrs.Seil --Info-- init 10.9
5/27/15 6:56:39.000 PM kernel[0]: org.pqrs.Seil --Info-- start
5/27/15 6:56:39.958 PM Seil[1062]: Invalid color System, labelColor (warning given only once)
5/27/15 6:56:40.185 PM Seil[1062]: checkForUpdates https://pqrs.org/osx/karabiner/files/seil-appcast.xml
5/27/15 6:56:40.187 PM Seil[1062]: Session state has been changed. (active)
5/27/15 6:56:40.188 PM Seil[1062]: observer_IONotification
5/27/15 6:56:40.188 PM Seil[1062]: [INFO] BRIDGE_USERCLIENT_SYNCHRONIZED_COMMUNICATION connection is null
5/27/15 6:56:40.447 PM Seil[1062]: Sparkle: ===== Seil.app =====
5/27/15 7:59:22.285 PM Seil[1062]: checkForUpdates https://pqrs.org/osx/karabiner/files/seil-appcast.xml
5/27/15 7:59:22.286 PM Seil[1062]: Sparkle: ===== Seil.app =====
5/27/15 7:59:22.314 PM Seil[1062]: unlockFocus called too many times. Called on <NSButton: 0x608000149f80>.
```

I will try a reboot right now and see what happens again.

tekezo, 2015-05-29:
Hmm, It's strange...
There is no problem in your log.

Is not there running Seil process in Activity Monitor even though such logs are in system.log?

justmytwospence, 2015-06-03:
Seil is running. Settings are not even persisted after I quit Seil. To be clear:
1. Set Seil to behave the way I want (bind caps lock to escape, for me)
2. Everything works as expected
3. Click quit in the Seil preferences
4. Reopen Seil
5. All my settings are now gone

tekezo, 2015-06-03:
Is this problem reproduced in the clean environment? (eg. Guest User)

closed 2015-08-06 by tekezo