pqrs-org/Seil #48: Seil cannot connect with kernel extension in OS X 10.11

dsdshcym, 2015-06-09:


mrexroad, 2015-06-09:
plus one

tekezo, 2015-06-09:
Thank you for feedback.
I'll release Seil for OS X 10.11.
Please wait for a while.

dsdshcym, 2015-06-09:
@tekezo 
Thank you!
Can't live without Seil :)

alexshenia, 2015-06-09:
11.1.3 woks fine for OS X 10.11 :+1:

cmsj, 2015-06-09:
I got a kernel panic after installing 11.1.3 and rebooting, but this being a beta OS, and the resulting log being very low on information, it's hard to say what the cause was. After the panic reboot, OSX came up fine and Seil works :)

tekezo, 2015-06-09:
I've released Seil 11.2.0 that works in OS X 10.11.
https://pqrs.org/osx/karabiner/seil.html.en

tekezo, 2015-06-09:
@cmsj
Thank you for feedback.
If the panic will be reproduced, let me know.

closed 2015-06-09 by tekezo

Jayphen, 2015-08-04:
I'm having this same problem as of the latest beta, running Seil 11.2.2. I tried uninstalling and reinstalling, but I still get the 'kernel extension' error.

tekezo, 2015-08-05:
OS X 10.11 beta6 fails to load kext automatically.
Please execute the following command in Terminal.app.

```
sudo /Applications/Seil.app/Contents/Library/bin/kextload load
```

Jayphen, 2015-08-05:
That works.

Is this something that has to be run at every startup?

tekezo, 2015-08-05:
Yes.
I don't understand whether it's an intended change of OS X 10.11 or a bug of OS X 10.11 yet.
If it is new OS X design, I'll fix Seil to we don't have to load kext by hand.

tekezo, 2015-08-08:
I've fixed the kext loading issue on OS X 10.11 beta 6 at Seil-v11.3.0.
Please upgrade Seil to the latest stable version.

alioguzhan, 2016-09-23:
Hey, same issue here with macOS Sierra. I have update to beta version [12.1.2] but nothing changed. Anything to do?

Sibicle, 2016-09-27:
Leaving this here, for anyone else encountering the same problem after updating to Sierra:

https://github.com/tekezo/Karabiner-Elements/blob/master/usage/README.md

@tekezo has a new project, "Karabiner Elements", that provides the functionality I achieved using Seil. Namely, remapping caps lock.