pqrs-org/Seil #49: Switching Capslock with ESC fails

ahhatem, 2015-06-18:
Hi,

I am having an issue replacing the Capslock key with ESC. Only escape doesn't work. Replacing with any other key works fine.

I am on macosx 10.10.3
Seil 11.2.0

Seil is on the login items.

Appreciate the support. 
Thanks
Ahmed

tekezo, 2015-06-18:
Did you change the escape key by Karabiner?
If so, the caps lock key is changed in Seil, then Karabiner.

Please disable Karabiner setting or use this way.
https://pqrs.org/osx/karabiner/faq.html.en#capslock

ahhatem, 2015-06-18:
That solves it. Many thanks.
May I ask if this is a behavior change. I don't think I had to do that before.

tekezo, 2015-06-18:
This behavior is not changed since the initial release.
I guess you did not use Karabiner setting or did that application key setting.

closed 2015-06-18 by ahhatem