pqrs-org/Seil #52: May not work with El Capitan

Grafz, 2015-07-12:
I've set the caps lock key to 53 and rebooted but it doesn't seem to work. (never had problems with Yosemite)

[image]

[image]

[image]

tekezo, 2015-07-13:
Are you using the latest stable version? (v11.2.0)
You can confirm Seil version in "Misc & Uninstall" tab.

Grafz, 2015-07-14:
Yup.

[image]

tekezo, 2015-07-14:
I cannot reproduce this issue.
Please reinstall Seil, then restart your system.

If your issue still happen, please confirm this issue in a clean environment like Guest User.

Grafz, 2015-07-16:
It seems to work now. I restarted in a different partition and when I came back to start tests in the El Capitan partition it seems to work. Sorry for what looks like a false alarm.

closed 2015-07-16 by Grafz