pqrs-org/Seil #53: Caps Lock key indicator

ericcong, 2015-08-02:
I mapped the Caps Lock key to F19 so that I can use it to change input source. Also I set it to "No Action" in OS X's Preference -> Keyboard -> Keyboard -> Modifier Keys.

It's working great and very convenient, but the Caps Lock indicator (the little green light on the Caps Lock key) is not working anymore. I think it might be good to have this indicator since it can tell me which input source I'm now using. 

Is there a way to make the Caps Lock key indicator back to work?

Thanks!

tekezo, 2015-08-03:
Karabiner does not support to use caps lock LED independently.
(Karabiner synchronize LED and caps lock state.)

You might can do that by maclight + input source observer.
https://github.com/busyloop/maclight

I think that hammerspoon might be input source observer if you want.
http://www.hammerspoon.org/

ericcong, 2015-08-03:
OK, thanks a lot!

closed 2015-08-06 by tekezo