pqrs-org/Seil #54: # - number sign key code

DataTables, 2015-09-17:
First - thank you for publishing this software! Making the caps lock useful is fantastic.

I wonder if you could add the number sign (#) to your list of key codes, or let us know what it is (or say how key codes can be found)?

I'm in the UK, so I need to use alt-3 to get a # and it really isn't the easiest key stroke (particularly with a broken wrist :-) ) and mapping it to the caps lock would be awesome.

tekezo, 2015-09-17:
You cannot specify a character directly in Seil.
You have to change your caps lock to alt-3 by the following procedure.

* Step 1: Change your caps lock to normal key such as f19 (keycode 80) by Seil.
* Step 2: Change f19 key to alt-3 by Karabiner. Use new setting by this private.xml.
https://github.com/tekezo/Files/blob/master/Karabiner/private.f19_to_alt_3/private.xml

How to use private.xml:
https://pqrs.org/osx/karabiner/document.html#privatexml

DataTables, 2015-09-17:
Got it - thanks!

closed 2015-09-17 by DataTables