pqrs-org/Seil #55: Opening Seil gives a tR

ethus3h, 2015-10-11:


tekezo, 2015-10-12:
I never faced such issue...
Please restart Seil. (Quit Seil from Activity Monitor and then open Seil from Launchpad.)

If your issue is still happen after restart, are there any messages in system log?
You can confirm logs from Console.app.
https://pqrs.org/osx/karabiner/faq.html.en#console

ethus3h, 2015-10-18:
Issue is still happening! I can't attach the log file, because "Attaching documents requires write permission to this repository. Try again with a PNG, GIF, or JPG." (WTF?)

Anyway, here it is pasted:
```
2015.10.18 18:11:44.000 kernel[0]: Sandbox: net.interacto.Fl(628) System Policy: allow(0) mach-priv-task-port 9753
2015.10.18 18:11:44.000 kernel[0]: com.apple.dock.e[9753] triggered unnest of range 0x7fff8b400000->0x7fff8b600000 of DYLD shared region in VM map 0x510bf113ade4c06b. While not abnormal for debuggers, this increases system memory footprint until the target exits.
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=4
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=5
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=6
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=7
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=8
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=9
2015.10.18 18:11:44.505 com.apple.dock.ecti[9753]: CUIThemeStore: No theme registered with id=10
2015.10.18 18:11:45.074 Seil[584]: *** Assertion failure in CGImageBlockSetRef __csiCompressImageProviderCopyImageBlockSetWithOptions(void *, CGImageProviderRef, CGRect, CGSize, CFDictionaryRef)(), /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreUI/CoreUI-362/CoreTheme/ThemeStorage/CUIThemeRendition.m:982
2015.10.18 18:11:45.077 Seil[584]: CGImageBlockSetRef __csiCompressImageProviderCopyImageBlockSetWithOptions(void *, CGImageProviderRef, CGRect, CGSize, CFDictionaryRef) unknown pixel format 0
2015.10.18 18:11:45.077 com.apple.xpc.launchd[1]: (Lastfm.restart[9759]) Service exited with abnormal code: 1
2015.10.18 18:11:45.078 com.apple.xpc.launchd[1]: (Lastfm.restart) Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
2015.10.18 18:11:45.088 Seil[584]: (
	0   CoreFoundation                      0x00007fff99b50bd2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff9b077dd4 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff99b558aa +[NSException raise:format:arguments:] + 106
	3   Foundation                          0x00007fff96d7eb1e -[NSAssertionHandler handleFailureInFunction:file:lineNumber:description:] + 169
	4   CoreUI                              0x00007fff8d6bf0bf __csiCompressImageProviderCopyImageBlockSetWithOptions + 562
	5   CoreGraphics                        0x00007fff89bf124b CGImageProviderCopyImageBlockSetWithOptions + 164
	6   CoreGraphics                        0x00007fff89c2c7af img_blocks_create + 523
	7   CoreGraphics                        0x00007fff89bf5919 img_data_lock + 1891
	8   CoreGraphics                        0x00007fff89bf42e3 CGSImageDataLock + 151
	9   libRIP.A.dylib                      0x00007fff89905234 ripc_AcquireImage + 972
	10  libRIP.A.dylib                      0x00007fff898f8d22 ripc_GetRenderingState + 2706
	11  libRIP.A.dylib                      0x00007fff898f8126 ripc_DrawRects + 98
	12  CoreGraphics                        0x00007fff89be26b7 CGContextFillRects + 107
	13  CoreGraphics                        0x00007fff89be2648 CGContextFillRect + 141
	14  QuartzCore                          0x00007fff94bfb0a3 _ZN2CA6Render12_GLOBAL__N_125create_image_by_renderingEP7CGImageP12CGColorSpacej + 910
	15  QuartzCore                          0x00007fff94bef62f _ZN2CA6Render12create_imageEP7CGImageP12CGColorSpacej + 2558
	16  QuartzCore                          0x00007fff94bee992 _ZN2CA6Render10copy_imageEP7CGImageP12CGColorSpacejd + 395
	17  QuartzCore                          0x00007fff94beb6c6 -[CALayer(CALayerPrivate) _copyRenderLayer:layerFlags:commitFlags:] + 786
	18  QuartzCore                          0x00007fff94beaf49 _ZN2CA7Context12commit_layerEPNS_5LayerEjjPv + 103
	19  QuartzCore                          0x00007fff94beae76 _ZN2CA5Layer16commit_if_neededEPNS_11TransactionEPFvPS0_jjPvES4_ + 330
	20  QuartzCore                          0x00007fff94be941e _ZN2CA7Context18commit_transactionEPNS_11TransactionE + 1766
	21  QuartzCore                          0x00007fff94be8a98 _ZN2CA11Transaction6commitEv + 508
	22  QuartzCore                          0x00007fff94bf428f _ZN2CA11Transaction17observer_callbackEP19__CFRunLoopObservermPv + 71
	23  CoreFoundation                      0x00007fff99ae5e07 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
	24  CoreFoundation                      0x00007fff99ae5d77 __CFRunLoopDoObservers + 391
	25  CoreFoundation                      0x00007fff99ac4d58 CFRunLoopRunSpecific + 328
	26  HIToolbox                           0x00007fff8b531d55 RunCurrentEventLoopInMode + 235
	27  HIToolbox                           0x00007fff8b531a97 ReceiveNextEventCommon + 184
	28  HIToolbox                           0x00007fff8b5319cf _BlockUntilNextEventMatchingListInModeWithFilter + 71
	29  AppKit                              0x00007fff88cd4f3a _DPSNextEvent + 1067
	30  AppKit                              0x00007fff88cd4369 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
	31  AppKit                              0x00007fff88cc8ecc -[NSApplication run] + 682
	32  AppKit                              0x00007fff88c92162 NSApplicationMain + 1176
[... cut ...]
2015.10.18 18:13:25.000 kernel[0]: ZFS: defer_leaf bad zap detected
2015.10.18 18:13:25.000 kernel[0]: ZFS: purgedir detected corruption. dropping 1097020
2015.10.18 18:13:25.514 com.apple.xpc.launchd[1]: (Lastfm.restart[9933]) Service exited with abnormal code: 1
2015.10.18 18:13:25.514 com.apple.xpc.launchd[1]: (Lastfm.restart) Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
2015.10.18 18:13:25.000 kernel[0]: ZFS: defer_leaf bad zap detected
2015.10.18 18:13:25.000 kernel[0]: ZFS: purgedir detected corruption. dropping 1138908
```

tekezo, 2015-10-19:
Thank you for log.

I guess these lines describe the cause.

Are you using custom window manager or using Quartz debug feature?
Does this issue happen in the clean environment such as Guest User?

```
2015.10.18 18:11:45.074 Seil[584]: *** Assertion failure in CGImageBlockSetRef __csiCompressImageProviderCopyImageBlockSetWithOptions(void *, CGImageProviderRef, CGRect, CGSize, CFDictionaryRef)(), /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreUI/CoreUI-362/CoreTheme/ThemeStorage/CUIThemeRendition.m:982
2015.10.18 18:11:45.077 Seil[584]: CGImageBlockSetRef __csiCompressImageProviderCopyImageBlockSetWithOptions(void *, CGImageProviderRef, CGRect, CGSize, CFDictionaryRef) unknown pixel format 0
```

ethus3h, 2016-10-25:
I'm sorry I've left this issue unreplied so long. I was not using a custom window manager or Quartz debug feature, as far as I know, but was using some tweaks like CDock and Bartender. Assuming the issue hasn't reappeared for other users (I'm no longer regularly using Mac OS), I'll close this issue. (I have no objection to other folks reopening it if they run into this.) Thanks. :)

closed 2016-10-25 by ethus3h