pqrs-org/Seil #56: Seil not working with OS X (10.11.1) El Capitan

arcseldon, 2015-11-10:
Hi, been using Seil successfully to switch caps lock for years, right up to Yosemite. Great app, and real convenience for VIM bindings. Recently downloaded OS X El Capitan, which I thought was a minor dot update to Yosemite, and it now appears that Seil caps lock switch to use Escape action is no longer working.  The symptoms are that Seil runs etc, and is setup to use key code 53 but nothing happens. Please could let me know if this is an issue because you don't yet support El Capitan, and whether you plan to provide such support.

tekezo, 2015-11-11:
Seil-11.3.0 works well with OS X 10.11.
Please confirm your Seil is the latest stable release.
https://pqrs.org/osx/karabiner/seil.html.en

If you are using Seil-11.3.0, please confirm "No Action" configuration.
(The system setting might be reset after OS upgrade.)
https://pqrs.org/osx/karabiner/seil.html.en#usage

arcseldon, 2015-11-11:
Hi @tekezo - thank you for your quick reply. Yes, I can confirm it now works. I was indeed on an older version. Curiously, I did have the settings set so that "Misc & Uninstall -> Check for updates on startup" was set to "The stable release only (default)"  Assuming therefore that 11.3.0 is not considered a stable release. Perhaps I should modify the default settings too.  All the same, great app, thank you very much for the assistance. Closing this as user error only.

closed 2015-11-11 by arcseldon