pqrs-org/Seil #57: Mapping caps lock to fn on an apple bluetooth keyboard.

TroyJ, 2015-11-19:
I'm trying to use caps-lock on my external keyboard as a modifier key, so I can control the cursor, etc.

I have it working on the local keyboard, but not on the external keyboard.  Why is this? Any workarounds?

My config is based on this issue ...
https://github.com/tekezo/Seil/issues/44
... as suggested I tried mapping caps-lock to f-19 using seil.

I then mapped fn+h to cursor-left.

<?xml version="1.0"?>
<root>
	<item>
		<name>Troy's stuff</name>
		<identifier>private.troy</identifier>
		<autogen>
			__KeyToKey__
			KeyCode::F19,
			KeyCode::FN
		</autogen>
		<autogen>
			__KeyToKey__
			KeyCode::H, ModifierFlag::FN,
			KeyCode::CURSOR_LEFT
		</autogen>
	</item>
</root>

tekezo, 2015-11-20:
It's a limitation of Seil.
You cannot change keys to fn in external keyboards.
Please change caps-lock to fn by using Seil and Karabiner together like yours.

TroyJ, 2015-11-20:
The goal is to press capslock and h, and have the cursor go left

I'm trying to use caps-lock on my external keyboard as a modifier key, so I can control the cursor, etc.

I tried fn, right control, also a virtual custom modifier key.

TroyJ, 2015-11-20:
The goal is to press capslock and h, and have the cursor go left (using external keyboard).

It doesn't have to be via the "fn" modifier. I tried fn, right-control, also a virtual custom modifier key... in each case it did not map on the external keyboard.

tekezo, 2015-11-20:
Is caps-lock in your external keyboard changed properly?

Please confirm the behavior by EventViewer.
https://pqrs.org/osx/karabiner/document.html.en#eventviewer

If caps-lock key sends the fn key down event and up event at the same time,
you should do "No Action" configuration for external keyboard.
https://pqrs.org/osx/karabiner/seil.html.en#usage

TroyJ, 2015-11-20:
Yes. It's changed properly, please see below. I just remapped it to CONTROL_R since we have issues with FN, still doesn't work.

```
<?xml version="1.0"?>
<root>
	<item>
		<name>Troy's stuff</name>
		<identifier>private.troy</identifier>
		<autogen>
			__KeyToKey__
			KeyCode::F19,
			KeyCode::CONTROL_R
		</autogen>
		<autogen>
			__KeyToKey__
			KeyCode::H, ModifierFlag::CONTROL_R,
			KeyCode::CURSOR_LEFT
		</autogen>
	</item>
</root>
```

Internal Keyboard:
```
eventType:FlagsChanged    code:0x3e       name:Control_R       flags:Ctrl                               misc:KeyCode::CONTROL_R	characters:    
eventType:FlagsChanged    code:0x3e       name:Control_R       flags:                                   misc:KeyCode::CONTROL_R	characters:    
eventType:KeyDown         code:0x7b       name:Left            flags:NumPad Fn                          misc:KeyCode::CURSOR_LEFT	characters:    
eventType:KeyUp           code:0x7b       name:Left            flags:NumPad Fn                          misc:KeyCode::CURSOR_LEFT	characters:    
```

External Keyboard:
```
eventType:FlagsChanged    code:0x3e       name:Control_R       flags:Ctrl                               misc:KeyCode::CONTROL_R	characters:    
eventType:FlagsChanged    code:0x3e       name:Control_R       flags:                                   misc:KeyCode::CONTROL_R	characters:    
eventType:KeyDown         code:0x4        name:h               flags:                                   misc:KeyCode::H	characters:h   
eventType:KeyUp           code:0x4        name:h               flags:                                   misc:KeyCode::H	characters:h   
```

tekezo, 2015-11-20:
OK.
Please confirm "No Action" configuration for the external keyboard.
https://pqrs.org/osx/karabiner/img/seil-noaction-multiple-devices@2x.png

TroyJ, 2015-11-20:
Confirmed. No action on caps-lock key only.

TroyJ, 2015-11-20:
Ah... I missed the dropdown. I disabled it for internal only, not external. That's fixed it. Thankyou for your help!

Is this the place to donate something?
https://pqrs.org/osx/karabiner/donation.html.en

tekezo, 2015-11-20:
> That's fixed it.

It's great!

> Is this the place to donate something?

Yes.

TroyJ, 2015-11-20:
Done! Thanks again.

closed 2015-11-20 by TroyJ

tekezo, 2015-11-20:
I confirmed your donation.
Thank you!

sdwebster, 2016-07-06:
Works for me, too. Thanks!