pqrs-org/Seil #59: make error

wangweijia, 2016-01-11:
Touch build/Release/Seil.kext
    cd /Users/apple/Desktop/capslock/Seil-master/src/core/kext/10.11
    /usr/bin/touch -c /Users/apple/Desktop/capslock/Seil-master/src/core/kext/10.11/build/Release/Seil.kext

** BUILD SUCCEEDED **

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C server
pod install --no-repo-update
make[4]: pod: No such file or directory
make[4]: *** [all] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [build] Error 2
make: *** [all] Error 2

wangweijia, 2016-01-11:
No `Podfile' found in the project directory.

tekezo, 2016-01-11:
The Podfile is `src/core/server/Podfile`.

The build script set the PATH environment variable.
https://github.com/tekezo/Seil/blob/master/make-package.sh#L4
You should set GEM_HOME if your `pod` command is not in `/usr/local/bin`.

wangweijia, 2016-01-12:
my `pod` is in `/usr/local/bin`
in `src/core/server/` , `pod install` by my self, and install success, but `make` error :::

Touch build/Release/Seil.kext
    cd /Users/apple/Desktop/test/Seil-master/src/core/kext/10.11
    /usr/bin/touch -c /Users/apple/Desktop/test/Seil-master/src/core/kext/10.11/build/Release/Seil.kext

** BUILD SUCCEEDED **

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C server
pod install --no-repo-update
make[4]: pod: No such file or directory
make[4]: *** [all] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all] Error 2
make[1]: *** [build] Error 2
make: *** [all] Error 2

what shoud i do next?

wangweijia, 2016-01-12:
in `src/core/server/` ,` pod install` by my self, and install success
annotation `PATH="/bin:/sbin:/usr/bin:/usr/sbin:$GEM_HOME/bin"; export PATH`   on make-package.sh

it make success...
~。~

tekezo, 2016-01-12:
Execute me.
I fixed the PATH to include /usr/local/bin.
6c6bb6ea4a4ccdeac30e39312ffd2a1a4b43c4dd

Please try the current master branch.

closed 2016-10-31 by tekezo