pqrs-org/Seil #61: Feature request: Trigger Seil options when plug in/out keyboard

aemonge, 2016-02-08:
Hi !!! 
I've switched the 'command' and 'option' key so I can 'command-tab' on my none-mac external keyboard normally. But when the external keyboard isn't present, I don't want that mapping to run. So It would be great if I didn't have to toggle those options each time I switch from external keyboard to the none external one.

I hope this isn't as hard as it sounds, you just have to create a shell input to the Seil commands, and trigger those on a keyboard-connected Event (from an external tool or like)

tekezo, 2016-02-09:
Please use Hammerspoon for this purpose.
http://www.hammerspoon.org/

Please execute Seil's command line interface from Hammerspoon.
https://pqrs.org/osx/karabiner/seil.html.en#commandlineinterface

aemonge, 2016-02-09:
Thank you so VERY much @tekezo

closed 2016-02-09 by aemonge