pqrs-org/Seil #62: Spotlight search removes first letter

frankfriberg, 2016-02-09:
I have this problem with Spotlight Search in El Capitan, and it is only there when i use Seil. When i open spotlight search and start typing, the first letter disappears...

I don't know if it is only me, but I experienced it before and after a clean install.

More info:
- I used Seil to change the caps lock key to F18
- I got F18 mapped as a global hotkey for the app Divvy
- I didn't experience it in Yosemite, but I'm not sure if I had version 12 either

tekezo, 2016-02-10:
I cannot reproduce this issue.
Is this issue reproducible in the clean environment such as Guest User?

frankfriberg, 2016-02-10:
Yes I was able to reproduce it as Guest User.
I also tried to map the caps lock to a different key with no success...

tekezo, 2016-02-11:
Thank you for confirmation.
I guess there are other cause such as sticky modifier is enabled or another app grabs the first letter.
How do you open Spotlight Search? Is the command-space shortcut?

closed 2016-10-31 by tekezo