pqrs-org/Seil #64: Microsoft Sculpt fn toggle

christianmagill, 2016-03-28:
The Microsoft Sculpt keyboard has a physical "fn" toggle that I detest.

I've managed to remap my capslock key to work as OSX's "fn" key, but I am wondering, is there any way to make the combination of capslock (fn) and F1-11 work as if the "fn" toggle was activated, while still maintaining standard F1-11 functionality without the capslock key being used?

Thanks for any help. Love Seil!

tekezo, 2016-03-29:
Do you want to use your physical capslock key as fn toggle?
Please follow this procedure.

1. Change capslock key to f19 key (keycode 80) in Seil.
2. Change f19 key to fn toggle in Karabiner with new setting by this private.xml:
https://github.com/tekezo/Files/blob/master/Karabiner/private.f19_to_fn_functional/private.xml
3. Disable other function key settings in Karabiner in order to avoid confliction with new setting.

How to use private.xml:
https://pqrs.org/osx/karabiner/document.html#privatexml

christianmagill, 2016-03-29:
Thanks for the reply,

Currently I've followed instructions elsewhere on the repository to set physical "application" key to "fn" key and physical "capslock" key to the application key.

This allows me to have "fn" key functionality. For instance I can double tap "capslock" (fn) to bring up dictation. If I'm understanding you correctly, "fn toggle" would be a separate beast and I couldn't simply use the "capslock" (fn) key as a modifier for the F# keys, which in a perfect world is what I'd be looking for.

What would you suggest?

tekezo, 2016-03-29:
How about this version?
https://github.com/tekezo/Files/blob/master/Karabiner/private.f19_to_fn_functional/private.xml#L10-L15

christianmagill, 2016-03-29:
Much thanks,

That does the trick!

closed 2016-10-31 by tekezo