pqrs-org/Seil #66: remapping capslock not working El Capitan

r6203, 2016-05-17:
Hi,

I'm trying to remap caps lock to escape; without any success.
I've set caps lock to "No Action" in OSX Keyboard settings and enabled the "Change the caps lock key" option in Seil. Whatever keycode I assign, it's not working.

Remapping the command and option key is working, though.

Seil version: 12.1.1
OSX: 10.11.1
Keyboard: Logitech Wave (USB)

Besides Karabiner, there's nothing running which may interfere with the keyboard.
I also tried quitting Karabiner and restarting my system.

Any advice?

tekezo, 2016-05-18:
Could you tell me your using kexts by this procedure?

* Open Terminal.app.

* Execute this command in Terminal.app
```
kextstat | grep -v com.apple
```

* Copy and paste the result.
```
Example:

Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
  136    0 0xffffff7f82ddc000 0x3000     0x3000     org.pqrs.driver.NoEjectDelay (7.1.3) 2B85A4DD-477A-397F-A6F3-1FD20A720FF5 <33 5 4 3 1>
  137    0 0xffffff7f82ddf000 0x29000    0x29000    org.pqrs.driver.Karabiner (10.18.40) 240E97D4-0E49-38C5-B5EC-FEDC0911CE15 <33 5 4 3 1>
  138    0 0xffffff7f82e08000 0x6000     0x6000     org.pqrs.driver.Seil (12.1.1) 8E861A5F-918F-368D-B3C6-0F704CD85B9D <33 5 4 3 1>
  139    3 0xffffff7f82e0e000 0x63000    0x63000    org.virtualbox.kext.VBoxDrv (5.0.20) 63689CCF-CA21-3900-B5C6-487E41A57997 <7 5 4 3 1>
  140    0 0xffffff7f82e71000 0x14000    0x14000    com.intel.kext.intelhaxm (6.0.1) 8FF2C637-0A5E-367E-B007-5B08655B1E8A <7 5 4 3 1>
  142    0 0xffffff7f82e85000 0x8000     0x8000     org.virtualbox.kext.VBoxUSB (5.0.20) F158578A-D7B0-3B85-B742-9E7867BCF948 <141 139 39 7 5 4 3 1>
  143    0 0xffffff7f82e8d000 0x5000     0x5000     org.virtualbox.kext.VBoxNetFlt (5.0.20) FF55D9DF-7F3D-3827-BE31-B319D57BBCF8 <139 7 5 4 3 1>
  144    0 0xffffff7f82e92000 0x6000     0x6000     org.virtualbox.kext.VBoxNetAdp (5.0.20) FEA9836D-EE06-392E-903E-9CDC607668D5 <139 5 4 1>
```

r6203, 2016-05-18:
Sure: 

```
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   14    0 0xffffff7f81202000 0x2000     0x2000     org.tgwbd.driver.NullCPUPowerManagement (1.0.0d2) 93D39664-55FB-9220-39AD-A436B33DB160 <7 4 3>
   19    0 0xffffff7f8188b000 0x19000    0x19000    org.netkas.driver.FakeSMC (1364) F9BDD56C-9FE4-345E-BD34-5706382AF0C9 <11 7 5 4 3 1>
   43    0 0xffffff7f81ebe000 0x1b000    0x1b000    com.insanelymac.driver.AppleIntelE1000e (3.1.0) 89207809-F285-FD56-82D2-2FA4D5C68C9C <42 12 5 4 3 1>
   61    0 0xffffff7f80efb000 0x46000    0x46000    com.Logitech.Control Center.HID Driver (3.9.1) no UUID <60 58 41 32 5 4 3>
  102    0 0xffffff7f80a21000 0x33000    0x33000    com.paragon-software.filesystems.ntfs (83.1.14) 6AE6F0DA-07C1-36CA-9F69-FAE8DF304376 <7 5 4 1>
  106    0 0xffffff7f83149000 0x29000    0x29000    org.pqrs.driver.Karabiner (10.18.0) E6E09313-87A6-30F6-94BE-61DA4EC19A8D <32 5 4 3 1>
  107    3 0xffffff7f83172000 0x60000    0x60000    org.virtualbox.kext.VBoxDrv (5.0.16) 35F02C29-8027-3636-ACD2-6F171A50C1D8 <7 5 4 3 1>
  108    0 0xffffff7f831d2000 0x6000     0x6000     org.pqrs.driver.Seil (12.1.1) 8E861A5F-918F-368D-B3C6-0F704CD85B9D <32 5 4 3 1>
  110    0 0xffffff7f831d8000 0x8000     0x8000     org.virtualbox.kext.VBoxUSB (5.0.16) BD2570CD-7993-3635-BE1E-FAB977FFBD5C <109 107 41 7 5 4 3 1>
  111    0 0xffffff7f831e0000 0x5000     0x5000     org.virtualbox.kext.VBoxNetFlt (5.0.16) 2C1775A7-B7A3-32A2-9E8B-951C1B499870 <107 7 5 4 3 1>
  112    0 0xffffff7f831e5000 0x6000     0x6000     org.virtualbox.kext.VBoxNetAdp (5.0.16) B108642D-2C95-3CB4-8D0B-01F867E60FBF <107 5 4 1>
```

Yes, I'm using Hackintosh...

r6203, 2016-05-18:
The Logitech Control Center caused the issue.
Remapping capslock is working fine now.

Thanks, @tekezo! 👍

closed 2016-05-18 by r6203

keeprock, 2017-03-20:
@r6203 Hi, I've got similar problem. Are you using any Voodoo drivers?

r6203, 2017-03-20:
@keeprock  To be honest, I don't know - I'm not using macOS anymore. Like I said above, the issue for me was using the Logitech Control Center.

keeprock, 2017-03-23:
@r6203 alright, thanks anyway.