pqrs-org/Seil #67: Option_R(61) に FN(63) を割り当てたいのですがうまくいきません

macoril, 2016-05-18:
【端末】MacBook Pro (Retina 13-inch、Early 2015)
【OS】OS X El Capitan 10.11.4

【困っていること】
ALT_R 61(OPTION_R) -> FN 63 の割り当てがうまくいかない

【経緯】
初めまして、いつも Karabiner に大変お世話になっています。
この度 Realforce を入手いたしまして、 MacBook Pro で 利用するために、
Seil を利用させていただくことになりました。

Karabiner の EventViewer で各キーのコードを確認し、
  WIN 55(COMMAND_L) <-> ALT_L 58(OPTION_L)
などの割り当てはできたのですが、
  ALT_R 61(OPTION_R) -> FN 63
の設定を行って  再起動を行いましたが、 ALT_R を押し下げても EventViewer に出力されず、 F1 などと組み合わせても反応がありません。
Karabiner は無設定状態で試したり、 [Fn+F1..F12 to Fcuntional Keys] にチェックを入れて試したりしてみました。
どうしたら設定できるか、ご助言いただければ幸いです。

tekezo, 2016-05-18:
https://github.com/tekezo/Seil/issues/44#issuecomment-67437382

macoril, 2016-05-18:
こんなに早くお返事いただけて驚いております！
外部キーボードではFnキーの設定は無理なのですね。

Karabinerの方で、 
OPTION_R を FN にする設定を有効にすることで目的を果たせました。

ありがとうございます。

closed 2016-05-18 by tekezo