pqrs-org/Seil #68: Neil doesn't load under 10.12

neilio, 2016-06-13:
I know this was released today, but Seil doesn't load its kernel extension under 10.12. I can't seem to get the kext to load manually as with previous updates - is there way to work around this issue for now?

notjosh, 2016-06-14:
If it helps you, I've done a little digging. tl;dr is kext loads fine, but doesn't do anything for me (I remap caps).

```
$ sudo kextload /Library/Application\ Support/org.pqrs/Seil/Seil.10.11.signed.kext
$ kextstat | grep -v com.apple
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
  140    0 0xffffff7f834d4000 0x2000     0x2000     com.nvidia.CUDA (1.1.0) 5AFE550D-6361-3897-912D-897C13FF6983 <4 1>
  146    0 0xffffff7f82b0d000 0x6000     0x6000     org.pqrs.driver.Seil (12.1.0) 9420A514-D737-3FA6-B008-CDEA735BC820 <32 5 4 3 1>
```

For future reference, this is on beta 1:

```
$ system_profiler SPSoftwareDataType
Software:

    System Software Overview:

      System Version: OS X 10.12 (16A201w)
      Kernel Version: Darwin 16.0.0
      ...
```

notjosh, 2016-06-14:
Also fwiw, seems like it's hanging onto something and refusing to terminate/unload:

```
$ sudo kextunload /Library/Application\ Support/org.pqrs/Seil/Seil.10.11.signed.kext
(kernel) Can't unload kext org.pqrs.driver.Seil; classes have instances:
(kernel)     Kext org.pqrs.driver.Seil class org_pqrs_driver_Seil has 1 instance.
Failed to unload org.pqrs.driver.Seil - (libkern/kext) kext is in use or retained (cannot unload).
```

I assume it used to terminate/unload just fine on 10.11 at least :)

It's too late at night for me to dig into this now, but I'll take a look tomorrow if you haven't had a chance!

tekezo, 2016-06-18:
macOS 10.12 changes of the generic keyboard and mouse drivers made a great impact on Karabiner and Seil.
We should make a large changes in Karabiner and Seil architecture.
There is not a workaround for this issue.

Please wait an update of Karabiner and Seil for macOS 10.12.
(It may take a long time.)

bosr, 2016-06-18:
Thank you teseko for the analysis and the hard work.  

Best regards!

ariporad, 2016-06-19:
Thanks so much @tekezo!

BjornMulder, 2016-06-22:
Is there still no way to remap caps to esc without seil?
I really depend on this mapping..
(Yes I have done the googling)

tekezo, 2016-06-23:
@BjornMulder Unfortunately, I don't know another way. Please use OS X 10.11 for a while.

BjornMulder, 2016-06-23:
@tekezo 
Alright, thanks for the response.

comex, 2016-06-29:
If anyone reading this is impatient like me, I found a hacky alternative way to map keys.  This only works with AppleEmbeddedKeyboard, which is most/all of Apple's own keyboards (run `ioreg | grep AppleEmbeddedKeyboard` to see if you have one).

You need to have System Integrity Protection disabled.  I think that's sufficient these days (used to require setting the `kext-dev-mode` boot argument but that doesn't seem to be used anymore, despite still showing up in kextcache's output).

Find the IOKit personality corresponding to your keyboard, which should be in either `/System/Library/Extensions/AppleHIDKeyboard.kext/Contents/Info.plist` or `/System/Library/Extensions/AppleUSBTopCase.kext/Contents/PlugIns/AppleUSBTCKeyEventDriver.kext/Contents/Info.plist`.  Identify the right one by comparing ProductID/VendorID to your keyboard's IDs, which you can find in the System Information app under USB (builtin keyboards) or Bluetooth (I think).  You'll need to convert from hex to decimal.

You should see something like:

            <key>IOClass</key>
            <string>AppleEmbeddedKeyboard</string>
            [..more stuff..]
            <key>ProductID</key>
            <integer>601</integer>
            <key>VendorID</key>
            <integer>1452</integer>

Edit the file (you need sudo), and put in anywhere:

            <key>KeyboardUsageMap</key>
            <dict>
                <key>0x00070039</key>
                <integer>0x00070029</integer>
            </dict>

note: 7,0x39 and 7,0x29 are the HID usage codes for caps lock and escape, respectively; you can look these up.  You can remap any keys; to remap multiple entries, repeat the two lines with 0x in them.

Make sure the file is owned by user `root` and group `wheel`.  If you edited it from within sudo this should be the case, but if you copied the file or something it might not be; in that case run `sudo chown root:wheel /whatever/path/to/Info.plist`.

Run `sudo touch /System/Library/Extensions && sudo kextcache -update-volume /`.  You should see something like this (depending on which kext you modified):

    kext-dev-mode allowing invalid signature -67030 0xFFFFFFFFFFFEFA2A for kext AppleUSBTCKeyEventDriver.kext
    kext-dev-mode allowing invalid signature -67030 0xFFFFFFFFFFFEFA2A for kext AppleUSBTopCase.kext

And reboot.  If it worked, your mappings should be in effect.  If it didn’t work, it’s probably because it decided not to load the driver (in which case it uses the generic HID keyboard driver and your function keys will be broken); you can try doing the touch/kextcache/reboot sequence again.

zeux, 2016-06-29:
@comex Thank you! Just want to confirm this works as written with SIP disabled and without any other actions required. To bind CapsLock via a shortcut, I remapped it to F13 (0x00070068) so that it's recognized as a valid key. FWIW on my MacBook the internal keyboard has a ProductID of 610 and is located in AppleUSBTopCase.kext, but I imagine this may be different between different models.

kennonb, 2016-07-03:
@BjornMulder If you're wanting to just re-map to ESC, you could use Keyboard Maestro. Ex. http://dsh.re/b3cd6 

However, doing more of the complex setup of using it primarily as a ~F19 key and then remapping that to CMD+CTRL+SHIFT+OPT isn't possible with Keyboard Maestro. At least not that I've been able to achieve.

vmax77, 2016-07-04:
@kennonb Does Keyboard Maestro work with Sierra then? And just to be clear, if I want to use remap Caps as F19 and use as my Alfred toggle - Keyboard Maestro would do the trick?

kennonb, 2016-07-05:
You won't be able to remap the key in that fashion unfortunately. At least not in a way that's really responsive. You could use AppleScript to try and pass key shortcuts back to Keyboard Maestro, but I found it to cause too much of a delay in processing, so I abandoned that idea.

BrendanThompson, 2016-07-17:
@tekezo any updates on an ETA for even a bleeding edge version of Seil :)

paul-howard, 2016-07-30:
For any fellow Sierra beta-testers looking to emulate the Seil/Karabiner "F19" trick to remap their CapsLock key to a vim-friendly "press-for-ctrl/tap-for-escape" configuration (as outlined nicely by @brendonrapp [here](http://www.economyofeffort.com/2014/08/11/beyond-ctrl-remap-make-that-caps-lock-key-useful/)), I was able to accomplish this by simply remapping CapsLock to Ctrl in System Prefs > Keyboard > Keyboard > Modifier Keys, then applying @kennonb's Keyboard Maestro example.

I'll still be waiting with bated breath for a Sierra-compatible version of Karabiner though, so I can get my universal Vi Mode back!

LoMiLiMiT, 2016-08-02:
@vmax77 @paul-howard This is my workaround atm with keyboard maestro: http://imgur.com/a/DhOzr
It works pretty good :) You can easily choose F19 or other F keys.

mistadikay, 2016-08-03:
@LoMiLiMiT worked like a charm. Thanks a ton!

nhynes, 2016-08-04:
Here's a [quick python script](https://gist.github.com/nhynes/7bcc80bbe5e005c8520abda053320519) that makes the changes as described by @comex.

Note that you will probably have to twiddle the xpath syntax a bit since I have only tested it on my own setup (macOS w/ SPI keyboard).

priezz, 2016-08-18:
Here https://stackoverflow.com/a/16125341 are the keycodes, that can be used with comex's approach.

sprig, 2016-09-03:
Does anyone know how to get the product IDs for @comex's approach on a macbook (late 2015)?

The built in keyboard doesn't show up under USB, nor does the microsoft sculpt keyboard. The latter shows up as "Wireless Transceiver" (and the dongle indeed is connected to a mouse as well).

Thanks in advance!

nklmilojevic, 2016-09-11:
Has anyone been able to simulate  ctrl+alt+cmd+shift (hyper) in Sierra? I updated today and I used in combination with BetterTouchTool to launch programs with capslock and a letter (for example Capslock + S for Spotify).

Does anyone have idea how to achieve that while I'm waiting for new versions of Seil and Karabiner?

Thanks

wxmtech, 2016-09-11:
App ain't working, and may take more time than it seems to be, so you know.

nklmilojevic, 2016-09-11:
Yeah, I know that, but I found some kind of a workaround with Keyboard Maestro and recording the keysequence in BTT.

asethwright, 2016-09-15:
[Karabiner Elements](https://github.com/tekezo/Karabiner-Elements) now has a build that is working for me fine on OSX 10.12. There is not a preferences pane yet so you have to modify the key via json config file. @tekezo has provided quick examples for typical mappings [here](https://github.com/tekezo/Karabiner-Elements/blob/master/usage/README.md).

wxmtech, 2016-09-16:
Incredible, working on karabiner-elements, anything that we could do to simulate two keys when pressing one key? I did not see an example like that.

vmax77, 2016-09-16:
@asethwright Hey, I cant seem to get Caps Lock to trigger F19 key. This is the json file I tried to use https://github.com/vmax77/Karabiner-Elements/blob/master/examples/change_caps_lock_to_f19.json. Am I doing something wrong?

wxmtech, 2016-09-16:
no comma after “fn”?

tseven, 2016-09-16:
Another alternative is using [Controller Mate](https://www.orderedbytes.com/controllermate/)
It is very powerful and will allow you to re-map your caps lock key.
[image]

I'm not sure what the limitations are for the unregistered version but it doesn't seem to affect the basic usage.

asethwright, 2016-09-16:
@wxmtech Sorry, I only successfully remapped my caps lock to escape. Maybe there is still a problem with f19 remapping.

BjornMulder, 2016-09-16:
Karabiner elements worked great! finally getting rid of my sore pinky syndrome. remapping caps-lock to escape works great!

LoMiLiMiT, 2016-09-16:
Does the "Hyper Key" work with the new Karabiner elements?

talha131, 2016-09-16:
@LoMiLiMiT Nope. Until [this issue](https://github.com/tekezo/Karabiner-Elements/issues/16) fixed.

LoMiLiMiT, 2016-09-16:
@talha131 ahhh ok damnit, tyvm. I hope the support comes soon :/ desperately waiting ^^

jooize, 2016-09-16:
Can Karabiner Elements do Caps Lock to Control/Escape? That and Hyper is what I most desire. ♥

wxmtech, 2016-09-17:
not to me, at least, there’s no example on hyper keys, now just swap keys.

I also love the launcher mode, damn, miss the whole suite.

wxmtech, 2016-09-17:
I see, thought the whole suite is completed in karabiner-elements.

andoriyu, 2016-09-17:
I don't want to sound like harsh, but could you please move discussion not related to *Seil* and this issue in question somewhere else?

wxmtech, 2016-09-17:
@andoriyu seil is not working in sierra, and only karabiner-elements can work related to caps lock swapping, what are you talking about, where to move the discussion. Next time @ someone.

steakknife, 2016-09-21:
Seil beta 12.1.0 appears to be working on Sierra.

zeekay, 2016-09-21:
@steakknife I'd double-check that assertion. Still broken here.

wxmtech, 2016-09-21:
newest beta still not working on second GM.

steakknife, 2016-09-21:
@zeekay When you're right, you're right. I don't use Seil for remapping Caps Lock anymore and just use Preferences to disable it.

jamessa, 2016-09-21:
I found my needs can be fulfill by just using Karabiner Elements and my remapped file is like this.

`
{
    "profiles": [
        {
            "name": "Default profile",
            "selected": true,
            "simple_modifications": {
                "caps_lock": "delete_or_backspace",
                "right_command": "left_control"
            }
        }
    ]
}
`

Thank you for the amazing work on Karabiner Elements. I think it's a right way to go.

LunarRed, 2016-09-22:
So I've cobbled to together a working Hyper Key hack in MacOS 10.12 Sierra using Karabiner-Elements and Keyboard Maestro. I posted my short walkthrough here: https://www.reddit.com/r/MacOS/comments/541oiq/how_to_get_back_your_hyper_key_in_macos_1012/

Hope this helps!

scottming, 2016-09-23:
Any update? Seil and Karabiner are more import than Siri for me. Should I have to back to OX EI or wait a few days? Could you give me a time answer? Thanks a lot, @tekezo!

talha131, 2016-09-23:
> Should I have to back to OX EI or wait a few days?

@scottming It depends on how you used Karabiner and Seil. If you just want to map one key to other (eg: Caps Lock to Fn, Control to Command) then Karabiner-Elements is ready.

But if you are looking for things like multiple key strokes as one (eg: Ctrl+j to left down arrow, left Control to Ctrl+Alt+Shift+Cmd), or multiple profiles then Karabiner-Elements does not have these features yet.

scottming, 2016-09-23:
@talha131 yes, I'am looking for things to make Capslock key as a Hyper key, just like that [my xml](https://gist.github.com/scottming/a9a756a9ed35467b18a4).

LunarRed, 2016-09-23:
@scottming Try my Karabiner-Elements and Keyboard Maestro hack/alternative. It's straightforward and will cover most of the Hyper Key shortcuts in your XML: https://www.reddit.com/r/MacOS/comments/541oiq/how_to_get_back_your_hyper_key_in_macos_1012/

diov, 2016-09-23:
For me, the problem is that I cant get the Device Vendor Id of my keyboard through EventViewer.
Also, all my config don't take effect.

davezatch, 2016-09-23:
@LunarRed Thanks! That covers 99% of my Hyper Key usage. A real life saver :)

LoMiLiMiT, 2016-09-25:
Does anyone know how to give one key 2 functions? Like hold Capslock does one thing and tap does another like f18 and f19 or so.

scottming, 2016-09-25:
@LoMiLiMiT My private.xml can do that.

LoMiLiMiT, 2016-09-25:
@scottming Could you maybe share it? And does it currently work under Karabiner-Elements? That would be really cool. Like having a key with 2 functions rather than having an extra key.

scottming, 2016-09-25:
@LoMiLiMiT Sorry, I just got up. I think this [private](https://gist.github.com/scottming/a9a756a9ed35467b18a4) is what you looking for. Capslock will send Esc when key is pressed alone or as a Hyper key when not alone.

[My article](http://scottming.com/2016/01/01/remap_keyboard/) is to explain that but written by Chinese, you could see the images to  understand that.

And here are more articles:

- [A Modern Space Cadet / Steve Losh](http://stevelosh.com/blog/2012/10/a-modern-space-cadet/#modern-software)
- [A useful Caps Lock key - BrettTerpstra.com](http://brettterpstra.com/2012/12/08/a-useful-caps-lock-key/)

wxmtech, 2016-09-26:
@scottming this is the pre Sierra era config file, the new upgraded system has issues with Karabiner and Seil. @LoMiLiMiT I suggest to wait for tekezo to release the config examples. Check this link https://github.com/tekezo/Karabiner-Elements <https://github.com/tekezo/Karabiner-Elements>.

The following is what’s going on with the next gen of karabiner(although no GUI config presented by now yet not hyper key support at this moment). Just some simple key swapping functionality with no hyper key support.

Karabiner GUI pref window can show, but it ain’t working. Seil has the problem launching for now.

LoMiLiMiT, 2016-09-26:
@scottming tyvm scott for sharing :) Sadly i have to wait a little bit more till it works for sierra uff :/ ty @wxmtech for pointing it out. Let's hope that this awesome tool soon works again.

arleym, 2016-09-28:
With the new MacOS keyboard drivers will it ever be possible to map capslock/F19 to multiple modifiers, or arrow keys? Or is there considerable limitation in what Karabiner Elements will be able to do compared to its predecessor on OS X?

I'm surprised how hard it's been without them! I'd hate to roll back to El Cap only to find out I was just impatient by a few days. I really appreciate what KE can do already!

LunarRed, 2016-09-28:
@arleym Try my Karabiner-Elements and Keyboard Maestro hack/alternative. It's straightforward and will cover most of the Hyper Key shortcuts (but not special keys like arrow and tab) in your XML: https://www.reddit.com/r/MacOS/comments/541oiq/how_to_get_back_your_hyper_key_in_macos_1012/

chenspc, 2016-09-28:
Not sure if anyone has reported on this, but I've just realised that the newest macOS Sierra (Version 10.12.1 Beta (16B2333a)) now allows you to map Caps Lock to Escape natively. That's pretty much all I needed Seil for in the first place, so the problem is sorted for me. 

I updated my system just now with my developer account, so this might not be available to the public yet. But hang in there.

asethwright, 2016-09-28:
@chenspc I understand this isn't available yet perhaps. But for when it is, can you how did you did this?

[46 more replies]