pqrs-org/Seil #70: Re-assigning § randomly stops working after sleep

yamalight, 2016-08-12:
I've re-assigned § to F13 to allow to use it inside of Electron-based apps, but when I put my mac to sleep and come back to it later - sometimes (but not every time) this assignment is removed until I re-open Seil, select the corresponding box (the value is still there) and hit enter. After that it starts working again until next sleep. 

I don't have this problem with re-assigned capslock - it stays assigned correctly whatever I do.
Any ideas what it might be? Tip how to debug that?

Using macOS 10.11.6 here.

yamalight, 2016-09-15:
Also seems to happen randomly on new apps launches.
I have very limited experience with system programming, but if you could point me to the relevant bit of code - I could try to solve it myself and send a PR.

tekezo, 2016-10-31:
Is there any crash report of Seil?
You can confirm it in Console app.

yamalight, 2016-10-31:
@tekezo unfortunately not. 
But I've since switched to Karabiner-Elements (because of macos update) - and that fixed it for me :)

tekezo, 2016-10-31:
It's great!

closed 2016-10-31 by tekezo