pqrs-org/Seil #72: Suggest a small update to users warning them of the issues with Sierra

digiltd, 2016-09-21:
I know it is not your fault or your responsibility to get it working. Unfortunately as you can see, so many people depend on your software. Thankfully I spotted the issues after I installed Sierra on my mac mini that just runs my media stuff. 

So I will be holding off updating my main development macs for now :)

But, it would be really useful if there was a update pushed to all users warning them that Karabiner and Seil are not yet working. I am sure many (as I would have) will think twice about updating the OS if they knew.

Thanks

nabn, 2016-09-22:
I hope I knew before the update 😔

m-meidani, 2016-09-22:
Please fix the compatibility issue with macOS Sierra! I cannot work any more!

nabn, 2016-09-22:
@m-meidani try this https://github.com/tekezo/Seil/issues/68#issuecomment-247415019
I got the caps lock key to change to escape and right option to control with this. 
Haven't tested extensively.

Also, I'm sure the author wants to get this working on macOS, but I also think that we should understand that it's not his "job" to fix it for us.

ZanderZhang, 2016-09-23:
mark

alioguzhan, 2016-09-23:
I had a kernel error after sierra update. Same error as #48

YA2O, 2016-09-26:
Same here... I cannot use my computer anymore... F**k you Apple!!!

akivascript, 2016-09-26:
C'mon, team. A lot of this commentary isn't helpful. I've been able to at least get the CAPS LOCK key doing double-duty as ESC and CTRL via Keyboard Maestro.

Sibicle, 2016-09-27:
Leaving this here, for anyone else encountering the same problem after updating to Sierra:

https://github.com/tekezo/Karabiner-Elements/blob/master/usage/README.md

@tekezo has a new project, "Karabiner Elements", that provides the functionality I achieved using Seil. Namely, remapping caps lock.

philc, 2017-10-31:
@tekezo please update the README of this project referring them to Karabiner Elements.